mennowo/TLCGen #50: Enkele kleine foutjes

WillemKinzel, 2019-04-03:
###hst.c:     
  char volgMaster = TRUE;  wordt 2x gegenereerd

Er wordt niet overal rekening gehouden met de eenheid van TX_PL_timer, TX_PL_max, TXA_PL, TXB_PL, TXC_PL, TXD_PL en TXE_PL.
Afhankelijk van #ifdef TX_PL_TE is dit seconden of tienden van seconden

halfstar.c:
  functie rhdhv_tx_change():
    In deze functie wordt 2 keer de functie check_signalplans() aangeroepen. Dit gebeurt echter elke programmarondje. 
    Als je dus een foute instelling hebt in een willekeurige signaalgroep (van bijvoorbeeld een tramrichting die geen 
    signaalplaninstellingen heeft, dan maakt de parser overuren.
    
###reg.c: 
  '#define NO_VLOG' staat op een plek nadat er aan NO_VLOG wordt gerefereerd
  
###tab.c:
  Dit wordt gegenereerd: IS_type[d02_1a] = DL_type | DKOP_type;
  DKOP_type is volgens de sysdef.c gedefinieerd als (DL_type+KOP_type), dus DL_type | DKOP_type is een beetje dubbelop. Logischer zou zijn:
  IS_type[d02_1a] = DKOP_type;, of
  IS_type[d02_1a] = DL_type | KOP_type;
  
###ov.c:
  #ifdef OV_ADDFILE
    PrioriteitsOpties_Add();
    PrioriteitsNiveau_Add();
  #endif
  De precompilerstatements #ifdef OV_ADDFILE en #endif moeten nog toegevoegd worden

mennowo, 2019-04-04:
Eerste punt is reeds verhoplen. De overige punten ga ik de komende tijd naar kijken. Halfstar wordt de komende weken sowieso aangepakt.

mennowo, 2019-04-17:
Het laatste punt is inmiddels ook verholpen

WillemKinzel, 2019-04-18:
Super!

Met vriendelijke groet,

Willem Kinzel
Adviseur Verkeersmanagement en -prognoses

T +31 (0)570 666 829  |  M +31 (0)6 11 62 15 85  |  E wkinzel@goudappel.nl<mailto:wkinzel@goudappel.nl>  |  [https://static.licdn.com/scds/common/u/images/logos/linkedin/logo_web_push_160x160_v1.png]   Willem Kinzel<www.linkedin.com/in/willemkinzel>  |  Twitter: @GoudappelTweets<https://twitter.com/goudappeltweets>

[cid:image005.png@01D3F8E2.D74B4CD0]
Goudappel Coffeng BV  |  Snipperlingsdijk 4  |  7417 BJ Deventer  |  Postbus 161  |  7400 AD Deventer  |  The Netherlands  |  www.goudappel.nl
Goudappel Coffeng is onderdeel van de Goudappel Groep. Wij werken vanuit Amsterdam, Den Haag, Deventer, Eindhoven, Leeuwarden en via partners in het buitenland

[cid:image004.png@01D492C2.8BBF7C80]
  Goudappel Coffeng en DAT.Mobility werkten mee aan deze wiki van CROW
  met databronnen en inwintechnieken voor beleid verkeer en vervoer. Klik hier<https://crow-verkeersonderzoek.nl/>

mennowo, 2019-08-05:
Het punt inzake de aanroep check_signalplans() heb ik nu als volgt aangepakt:
- er wordt binnen de functie set_tx_change() een vlag opgezet indien er een wijziging wordt doorgevoerd
- check_signalplans() wordt alleen aangeroepen indien de vlag waar wordt.
- er volgen dus nog steeds telkens weer aanroepen van check_signalplans, echter niet meer elke machineslag.
- Ik zet op de todo lijst om te bekijken/overwegen of de aanroepen naar check_sp() niet weg kunnen, bv door bij start regelen en doorvoeren wijzigingen eenmalig voor 2e realisaties (waar dit om gaat) ook direct de juistheid te bekijken. punt is natuurlijk dat je de betreffende prms altijd kunt wijzigen, waarna er dan geen check meer zou zijn.

Vwb de eenheid van TX_PL_timer etc.: wanneer je aangeeft waar de code hierop moet worden aangepast, voer ik dat graag door. Dat kan in een mail of (nog beter) door een pull-request.

De define NO_VLOG is verplaatst nu.

DL_type is verwijderd nu.

closed 2019-08-05 by mennowo