mennowo/TLCGen #55: Diverse issues

WillemKinzel, 2019-06-21:
Ik ben bij het gebruik van TLCGen de volgende puntjes tegengekomen:

###reg.c
    /* Bij nalopen op EG mag de volgrichting niet RR en FM
       gestuurd worden indien de voedende richting groen is */
    if (!R[fc29] || TNL[fc26]) { RR[fc26] &= ~BIT5; FM[fc26] &= ~BIT5; }
    Dit blokje moet onder de FM-instructies staan in plaats van erboven

halfstar.c:
    if (mlx == -1) for (ml = 0; ml < MLAMAX; ++ml) if (PRMLA[ml][fc]) { mlx = MLA; break; prml = PRMLA; mlmax = MLAMAX; }
    Het break-commando moet volgens mij pas na 'mlmax = MLAMAX;'
    
De parameters prmaltghst## worden niet gebruikt

Er zijn 2 verschillende functies met bijna dezelfde naam: Alternatief_halfstar en alternatief_halfstar. Het enige verschil is de eerste letter: 'A' dan wel 'a'
Dit kan verwarrend zijn.

In ###reg.add, ###hst.add en extra_func.c zijn functie-definities van functies zonder argumenten in de aanroep, zoals 
void KlokPerioden_Add()
{
}
Sommige compilers geven dan een warning. Met onderstaande wijziging kan je dat voorkomen:
void KlokPerioden_Add(void)
{
}

De parameter prmvolgmasterpl heeft een waarde (65.535) die veel hoger is dan de maximaal instelbare waarde in de parser (30.000/32.767). 
Dit treedt op in regelingen met de halfstarre structuur (slave-regelingen).

Bij gebruikt van intergroentijden zou het handig zijn als bij het genereren van de projectfile ook de preprocessor definitie CCOLTIG wordt gegenereerd.  

De geheugenelementen mmk## t.b.v. de dynamsche hiaattijden worden ten onrechte niet altijd gegenereerd.

Bij genereren van een projectfile voor ene regeling met halfstar en intergroentijden, wordt tigfunc.lib in de projectfile gezet i.p.v. trigfunc.lib.

Er worden soms koppelsignalen dubbel gebruikt. Dit betreft regelingen met de halfstarre structuur (master-regeling) en een uitgaande pelotonkoppeling.
In dat geval worden de uitgaande koppelsignalen 1 en 2 gebruikt door zowel de halfstarre module als de pelotonkoppeling. 
Er is volgens mij ook niet een ander koppelsignaal te kiezen in de GUI van TLCGen, wat wel het geval is bij een inkomende pelotonkoppeling.

mennowo, 2019-06-25:
Ha Willem, dank voor de feedback. Morgen komt er nog een nieuwe versie met wat fixes.
- FM > gefixt op Github
- haflstar.c > gefixt
- prms altghst > deze worden wellicht ten onrechte aangemaakt. Ga ik na
- naamgeving functie hst_alt > eens. zet ik op de lijst
- () en (void) > gefixt
- prmvolgmasterpl > waarde 65535 betekent 15 bits op. de parser accepteert waarschijnlijk slechts 32767 omdat het een signed short is. zet ik op de lijst om eens te checken
- CCOLTIG en trigfunc.lib in projectifle > is bekend: staat op de lijst
- geheugen elems mmk## > kun je specificeren wat "niet altijd" is? **hoe is dit te reproduceren?**
- koppelsignalen > op github reeds enige fixes hiervoor van de afgelopen dagen. er komt tzt ook een wiki pagina. de signalen beginnen binnen TLCGen nu altijd bij 1. Hij telt inmiddels per kruising vanaf 1.

WillemKinzel, 2019-06-25:
Beste Menno,

Bedankt voor je snelle acties!

De geheugenelementen mmk## t.b.v. de dynamische hiaattijden worden in de bijgevoegde tlc-file van k175 niet gegenereerd, maar in die van k176 wel, terwijl de boel toch redelijk gelijk is ingevuld.

Met vriendelijke groet,

Willem Kinzel
Adviseur Verkeersmanagement en -prognoses

T +31 (0)570 666 829  |  M +31 (0)6 11 62 15 85  |  E wkinzel@goudappel.nl<mailto:wkinzel@goudappel.nl>  |  [https://static.licdn.com/scds/common/u/images/logos/linkedin/logo_web_push_160x160_v1.png]   Willem Kinzel<www.linkedin.com/in/willemkinzel>  |  Twitter: @GoudappelTweets<https://twitter.com/goudappeltweets>

[cid:image005.png@01D3F8E2.D74B4CD0]
Goudappel Coffeng BV  |  Snipperlingsdijk 4  |  7417 BJ Deventer  |  Postbus 161  |  7400 AD Deventer  |  The Netherlands  |  www.goudappel.nl
Goudappel Coffeng is onderdeel van de Goudappel Groep. Wij werken vanuit Amsterdam, Den Haag, Deventer, Eindhoven, Leeuwarden en via partners in het buitenland

mennowo, 2019-08-05:
Inmiddels zit er in de versie op Github/appVeyor een nieuwe manier van omgang met koppelsignalen: zie tabblad Specials>PTP.
tigfunc/trigfunc is inmiddels ook verholpen.
Speelt het issue met mmk nog? Zo ja, stuur dan even direct een mail, of post het attachment op Github (niet via de mail). Het is in de eerdere mail weggevallen.

mennowo, 2019-08-05:
prmaltghst is inmiddels verwijderd, want deze wordt niet benut.

WillemKinzel, 2019-08-06:
Beste Menno,

Bedankt voor alle verbeteringen die je al gedaan hebt. Het issue met mmk## speelt nog wel. Als je met bijgevoegde tlc-file een CCOL-regeling genereert, kan je zien dat in de tab- en sys-file de geheugenelementen ontbreken.
Je andere vraag, over de eenheden van de signaalplanparameters, beantwoord ik later nog.

Met vriendelijke groet,

Willem Kinzel
Adviseur Verkeersmanagement en -prognoses

T +31 (0)570 666 829  |  M +31 (0)6 11 62 15 85  |  E wkinzel@goudappel.nl<mailto:wkinzel@goudappel.nl>  |  [cid:image001.jpg@01D54C6B.D03714A0]   Willem Kinzel<www.linkedin.com/in/willemkinzel>  |  Twitter: @GoudappelTweets<https://twitter.com/goudappeltweets>

[cid:image002.png@01D54C6B.D03714A0]
Goudappel Coffeng BV  |  Snipperlingsdijk 4  |  7417 BJ Deventer  |  Postbus 161  |  7400 AD Deventer  |  The Netherlands  |  www.goudappel.nl
Goudappel Coffeng is onderdeel van de Goudappel Groep. Wij werken vanuit Amsterdam, Den Haag, Deventer, Eindhoven, Leeuwarden en via partners in het buitenland

[Afbeelding met weg, scène, buiten, vrachtwagen  Automatisch gegenereerde beschrijving]<https://www.goudappel.nl/actueel/voorbereid-op-weg-naar-de-vrachtwagenheffing/>

wimpruijsers, 2022-12-09:
Is dit al verwerkt??

mennowo, 2024-10-23:
@WillemKinzel zijn er nog punten van dit issue actueel? Zo ja, kun je dan _per relevant item_ (!!) een nieuwe issue aanmaken? Dan zal ik dat bekijken, in samenspraak met de kerngroep van TLCGen. Dit issue sluit ik nu af, want het is zo oud dat het inmiddels niet meer gaat over de actuele stand van de generator.

closed 2024-10-23 by mennowo