mennowo/TLCGen #66: Alleen reset fasebewakingstijd bij start en einde hulpdienst

ccappendijk, 2024-12-04:
Momenteel wordt de fasebewakingstijd gereset tijdens het aanwezig zijn van een hulpdienst. Dit willen wij wijzigen naar de start of einde van een hulpdienst. Hiervoor is de prio.c aangepast. Deze nieuwe prio.c is doorgevoerd bij de K123456 regeling.

mennowo, 2024-12-10:
Deze fix geeft elders problemen:

Het gaat nu mis bij de aansturing van de BUS melding bij wachttijdvoorspellers: deze test op `!(RTFB & PRIO_RTFB_BIT)`.
Bij een lopende HD aanwezigheid is RTBF na de recente aanpassing niet waar (dus de uitkomst van de test is !(niet waar) = waar, waardoor de BUS melding ook tijdens HD aanwezigheid zal worden aangestuurd.

ccappendijk, 2025-01-10:
RTFB in reg.c vervangen door nieuwe boolv HDAANWEZIG. 
HDAANWEZIG wordt aangemaakt in prio.h en op- en afgezet in prio.c als er een hulpdienst is.
Code doorgevoerd op GIT K123456

LexTrafico, 2025-01-13:
Uit eigen ervaring: bij HDaanwezig beter niet kijken naar aanwezig zijn van hulpelement, maar naar de counter (C[cvchd##]).
Fraai zou zijn wanneer deze counters ook beschikbaar komen onder PrioInstellingen() in ###prio.c, bijvoorbeeld in de vorm 
iC_PRIOix[prioFC02bus] = cvc02bus;
zodat je ze in een loopje 0 ... prioFCMAX kan langslopen.

ccappendijk, 2025-01-14:
wij loopen idd (prio.c): 
------------------------------------------------------------------------------------
    HDAANWEZIG = FALSE;
    for (prio = 0; prio < prioFCMAX; ++prio)
    {
        if (iPrioriteit[prio] && iPrioriteitsOpties[prio] & poBijzonderRealiseren)
        {
            fc = iFC_PRIOix[prio];
            TegenHoudenStartGroen(fc, iStartGroen[prio]);
            if (iPrioriteitsOpties[prio] & poNoodDienst)
            {
               HDAANWEZIG = TRUE;
            }

        }
    }

closed 2025-01-14 by mennowo