mennowo/TLCGen #68: Twee compileerfouten bij Halfstar

ccappendijk, 2024-12-05:
1. Compileer fout is omdat in de sys.h wat ontbreken: 
ussyncok moet zijn usinK083syncok
uspervar moet zijn usinK081pervar
usperarh moet zijn usinK081perarh
2. Als er geen interne koppelingen zijn dan is er een compileerfout. Stukje code daarvoor moet worden aangepast. 

Ik heb Ronald gevraagd de betreffende .tlc op te sturen zodat we kunnen testen dat het dan is opgelost. Ronaldstuurt de .tlc file door.

ccappendijk, 2024-12-06:
[image]

ccappendijk, 2024-12-06:
[K082.txt](https://github.com/user-attachments/files/18036172/K082.txt)

Hernoemd naar .txt de .tlc file dan

LexTrafico, 2024-12-06:
In mijn herinnering treedt deze bug niet op wanneer je een kruispuntnummer van 5 karakters gebruikt als koppeling ID bij de koppeling instellingen (dus K0083 ipv K083).

mennowo, 2024-12-06:
Een interessante bug. Om in het verleden ontstane maar nu verloren gegane redenen wordt [hier](https://github.com/mennowo/TLCGen/blob/a21399f4de350a7fca8dc070e0949b8aa1222bbb/TLCGen.PluggedInItems/TLCGen.Generators.CCOL/CodeGeneration/HelperClasses/CCOLCodeHelper.cs#L13) de kruispunt naam die op die plek in de element naam zit er weer uitgehaald.

Dit vergt enig onderzoek naar het doel van de betreffende code, en waar dit wel/niet correct is om zo te doen.

mennowo, 2024-12-06:
Omtrent het tweede punt, dit betreft [deze code](https://github.com/mennowo/TLCGen/blob/025958adb11ef2ad42631f323d89d0a079cc3876/TLCGen.PluggedInItems/TLCGen.Generators.CCOL/CodeGeneration/Functionality/HalfstarCodeGenerator.cs#L466) waar gekeken wordt naar nalopen, wat indien die er niet zijn deze code oplevert:

    /* Afzetten IH[homschtegenh] */
    if (!IH[hkpact] && !IH[hpervar] && !SCH[schvar] && !IH[hplhd])
    {
        if (        )
        {
            IH[homschtegenh] = FALSE;
        }
    }

@ccappendijk Kan dit blok dan volledig weg indien er geen nalopen zijn, of blijft het aanwezig, maar wordt `homschtegenh` op `FALSE` gezet enkel obv het eerste `if` statement?

ccappendijk, 2024-12-06:
> Omtrent het tweede punt, dit betreft [deze [...]

De code mag dan worden: 
 /* Afzetten IH[homschtegenh] */
 if (!IH[hkpact] && !IH[hpervar] && !SCH[schvar] && !IH[hplhd])
 {
         IH[homschtegenh] = FALSE;
 }

closed 2024-12-09 by mennowo