mennowo/TLCGen #71: #include realfunc.c onder #control.c in reg.c (staat doorgevoerd bij k123456)

ccappendijk, 2024-12-15:
[image]

mennowo, 2024-12-17:
Wat is hier dan de eigenlijke fout? De volgorde van dergelijke includes zou eigenlijk niet zoveel uit moeten maken. De `#define REALFUNC` is sowieso al naar bovenin de reg.c verplaatst.

ccappendijk, 2024-12-17:
Waarschijnlijk omdat maatgevend_groen in de realfunc.c wordt aangemaakt en in de extra_func wordt aangeroepen. Dan moet realfunc erboven staan toch ? Een andere oplossing vind ik ook goed natuurlijk.

WillemKinzel, 2024-12-17:
Je kan ook een declaratie toevoegen van maatgevend_groen() in het bestand extra_func.h. Zoiets:

#if defined(REALFUNC)
    boolv Maatgevend_Groen(count fc);
#endif

Met vriendelijke groet,

Willem Kinzel
Adviseur Verkeersmanagement en -Prognoses

M +31 (0) 6 11 621 585
T  +31 (0) 570 666 222 (algemeen)
www.goudappel.nl<https://www.goudappel.nl/>
***@***.***<http://www.goudappel.nl/>

Blijf op de hoogte van het laatste mobiliteitsnieuws.
Schrijf u in voor onze nieuwsbrief.<https://www.goudappel.nl/nl/nieuwsbrief>

Goudappel | Postbus 161 | 7400 AD Deventer
Wij werken vanuit Amsterdam, Den Haag, Deventer,
Eindhoven, Leeuwarden en via partners in het buitenland

ccappendijk, 2025-01-09:
Zie email conversatie. Twee mogelijke oplossingen. Nu is er een compileerfout.

closed 2025-01-09 by mennowo