mennowo/TLCGen #86: WTV en HD (knipperen aftel LEDjes)

LexTrafico, 2025-02-11:
Het lijkt erop dat in TLCGen hulpdiensten bij aanwezigheid van een (actieve) wachttijdvoorspeller ook prioriteit krijgen wanneer het getoonde aantal aftel LEDjes onder de ingestelde ondergrens staat.
Dan wordt echter (omdát het aantal LEDjes onder de ondergrens zit) rr_twacht[] niet op gezet, waardoor het laatste LEDje niet gaat knipperen.
Zie hieronder. de wtv bij 22 staat hoog genoeg en gaat wel knipperen, die bij 24 niet.

[image]

Voorstel:  reg.c aanpassen onder "aansturing wachttijd lantaarns (niet tijdens fixatie of prio ingreep)" door ook te kijken naar !bHDAanwezig.

bestaand:
`if (!MM[mwtv22] || MM[mwtv22] >= PRM[prmwtvnhaltmax] || MM[mwtv22] <= PRM[prmwtvnhaltmin]) rr_twacht[fc22] = 0;`

voorstel:
`if (!MM[mwtv22] || MM[mwtv22] >= PRM[prmwtvnhaltmax] || (MM[mwtv22] <= PRM[prmwtvnhaltmin]) && !bHDAanwezig) rr_twacht[fc22] = 0;`

LexTrafico, 2025-02-11:
En dan volgens mij in ###prio.c ook hier

[image]

en onder PostAfhandelingPrio() hier
[image]

mennowo, 2025-05-26:
Hierover met Cyril gesproken: bedoeling is eigenlijk dat bij # leds < PRM, niet tegenhouden, óók niet door een hulpdienst. In plaats van aanpassen van de knipperaansturing zou dus wenselijker zijn te zorgen dat de beoogde functionaliteit gewoon gaat werken: nooit tegenhouden wtv-richting als # leds < PRM.

LexTrafico, 2025-08-05:
Is ook prima ;-)

mennowo, 2026-10-07:
@ccappendijk Is al duidelijk wat hiertoe moet worden aangepast?

mennowo, 2026-11-06:
Gekozen voor de oorspronkelijke oplossing: wél laten knipperen wachttijdvoorspeller ook als het aantal leds onder het ingestelde minimum zit in geval er een hulpdienst aanwezig is.

Wel verlenen van prioriteit in dit scenario kom in de backlog.

closed 2026-11-06 by mennowo