mennowo/TLCGen #88: Spec. generator doet het niet in de laatste versie; als je hem vanaf GIT zelf compileert wel. Even kijken waar dat aan zou kunnen liggen.

ccappendijk, 2025-02-25:


ccappendijk, 2025-02-25:
Dit geldt alleen voor de geinstalleerde versie bij mij. Als ik de portable van de website download gaat het wel goed.

mennowo, 2025-03-01:
> Het zou mogelijk een oplossing kunnen zijn als je na regel 84 toevoegt `fileStream.Close();`

Eigenlijk zou dat door de `using` vanzelf moeten gaan, want zodra `fileStream` buiten de scope komt wordt ie dan opgeruimd, en wordt de stroom gesloten.

@ccappendijk Kun je "doet het niet" wat nader specificeren? Is er een error? Zo ja welke?

In geval er geen error is, is dat eigenlijk an sich een fout; dan is er blijkbaar een catch-all clause ergens zonder melding, dat is voor productie niet handig.

Wat ook zou kunnen, en eigenlijk waarschijnlijker is, is dat er in de setup een bepaalde dependency niet wordt mee verpakt, die in de portable versie wel mee komt.

mennowo, 2025-03-12:
dit komt door een ontbrekende DLL in de setup

closed 2025-03-12 by mennowo