mennowo/TLCGen #95: Keuze voor aansturing waitsignaling op basis van aanvraag of detectiemelding schakelbaar

ccappendijk, 2025-03-11:
De keuze voor aansturing waitsignalering moet je nu opgeven in de TLCGen en dan wordt er stukje code gemaakt. Dat willen we graag schakelbaar dus beide stukjes code worden opgenomen afhankelijk van de keuze van de schakelaar. Naamgeving schakelaar: schtypeuswt

mennowo, 2025-03-12:
komt dit ipv de huidige checkbox "Wachtlicht" ?

ccappendijk, 2025-03-12:
Die blijft; geen aanpassing GUI. 
Als de checkbox zegt: Drukknopgebruik dan heeft de schakelaar de isntelling 1.
Als de checkbox zegt Aanvraag gezet dan heeft de schakelaar de instelling 0. 

De code die de TLCGen moet maken is dan: 
if (schakelaar) 
{ 
  <<hier de code voor drukknopgebruik die al wordt aangemaakt in de TLCGen>>
}
else
{
  <<hier de code voor aanvraag gezet die al wordt aangemaakt in de TLCGen>>
}

mennowo, 2025-03-12:
Gegeven de huidige UI

[image]

welke code moet onder welke schakelaar?

mennowo, 2025-03-12:
ah wacht je doelt op de algemene instelling voor type aansturing. nu kan ik het volgen