mennowo/TLCGen #100: Wachtstandgroen uitbreiden

ccappendijk, 2025-03-20:
Schakelaar WG$$ wijzigen naar parameter en volgende functionaliteit toevoegen. Betreft dus ook GUI aanpassing.

Toestand van het groen als er geen conflictaanvragen zijn. De lengte van dit groen is niet afhankelijk van verkeer op de betreffende richting of van een tijdsinstelling. Wachtgroen kan worden in- of uit geschakeld voor de fasecycli waarvoor deze optie gedefinieerd is.

Door middel van een parameter kan worden aangeven welke vorm van wachtstand gewenst is (PRM WGxx, default 0).

De waarde 1 houdt in dat de betreffende fasecyclus in wachtgroen blijft staan als deze op basis van een aanvraag groen is geworden. 

De waarde 2 houdt in dat de wachtstandaanvraag wordt gegenereerd wanneer er geen conflicterende fasecycli actief zijn. De waarde 0 houdt in dat de betreffende fasecyclus geen wachtstand groen heeft.

Cyril maakt nog code en zal deze opnemen op K123456

mennowo, 2025-05-05:
De nieuwe opzet is mogelijk verwarrend omdat er dan een SCH is voor aan/uit én een PRM die óók zorgt voor niet toepassen wachtstand in geval PRM == 0. Implementatie dus heroverwegen.

ccappendijk, 2025-05-06:
Ik heb overlegd met een aantal gebruikers en ook indiener van de vraag. Voor nu willen we toch vasthouden aan de schakelaar. De argumenten die ik hoor waren: gelijk aan werkwijze schym/prmymtype, uniforme invoer in regelscenario's (uit kan gewoon met een schakelaar), traffick werkt ook met deze schakelaar (niet met een parameter), diverse wegbeheerders zijn gewend met de schakelaar. Helemaal volledig zullen we het ook in het volgende kerngroepoverleg beslissen maar ik wil nu versie 12.4.0.14 laten uitbrengen met wat aanpassingen. De spec. generator moet nog wel mee bij deze aanpassing (@mennowo) . Ik stuur Menno nog aantal wijzigingen aan de huidige implementatie via K123456 (@mennowo).

LexTrafico, 2025-05-15:
Lokkertje: wanneer de suggestie uit de vorige comment doorgevoerd wordt, wil ik deze keer de specgen wel updaten ;-) .

ccappendijk, 2025-05-15:
In traffick2tlcgen is het nu een schakelaar (waarschijnlijk door de integratie met TLCGen). 

[image]

Ik wil er ook niet te lang bij stil blijven staan want wil zsm de ISG opnemen in de TLCGen (dat kan nadat deze 12.4.0.14 klaar is). Voor nu even zoals is voorgesteld wat mij betreft. Het is wat mij betreft ook geen halszaak maar nu alleen een uitbreiding van functionaliteit (op verzoek van Breda). Menno doet een uitbreiding van de spec. generator heb ik afgesproken ! Desgewenst bespreken we het in de kerngroep wat we willen met zijn allen. Ik bel je nog even.

mennowo, 2025-05-16:
Dit zit er nu in als volgt:

- SCH = zoals eerder
- PRM = _inclusief_ optie 0 voor 'geen wachtstand'
- Specgen ook geüpdatet, met extra paragraaf met tabel a la meeverlengen
- Defaults aangepast: default voor PRM = 2 (wg+a)
- Voor bestaande regelingen zorgen dat PRM wordt ingesteld op 2 voor richtingen met WG != nooit
- Als je een fc wijzigt van WG == nooit naar WG != nooit, instellen PRM op 2 indien die 0 was

Hiermee lijkt de implementatie volledig, en wordt ook rekening gehouden met bestaande regelingen en gebruikers die verder willen zonder functionele wijziging (en die de PRM mogelijk helemaal niet op zullen merken; die staat dus zo default netjes op 2).

Merk op: bij gebruik van afwijkende defaults, zullen die moeten worden aangepast, zodat de juiste WG type prm wordt meegenomen. Daar zal deze wel op 0 komen te staan, omdat die niet in de XML data zit, resulterend in de default waarde (0).

closed 2025-05-16 by mennowo

LexTrafico, 2025-05-19:
TOP, dank!