mennowo/TLCGen #116: Overlapgroen

ccappendijk, 2025-04-25:
Hoi Cyril, over onderstaande sprak ik je laatst. Bij deelconflicten willen we ook dat het langzaam verkeer een minimale tijd groen heeft op het moment dat de autorichting groen krijgt; wij noemen dit overlapgroen. Je zou dit in TLCGen op moeten kunnen geven in de matrix. In de ATB zit dat ook.

Ik heb hiervoor een functie gemaakt voor TLCGen.

 

Toegevoegd in tab.add:

 

  T_code[tog2205] = "og2205";  T_max[tog2205] = 5;  T_type[tog2205]  = TE_type; /* deelconflict overlapgroen 22 op 05 */

  Etc.

 

Toegevoegd in reg.add:

 

void Synchronisaties_Add(void)

{

dc_overlapgroen(fc22, fc05, tog2205, fc11, tog2211, SCH[schog2205]);

dc_overlapgroen(fc24, fc08, tog2408, NG  , NG     , SCH[schog2408]);

dc_overlapgroen(fc26, fc05, tog2605, fc11, tog2611), SCH[schog2605];

dc_overlapgroen(fc28, fc02, tog2802, NG  , NG     , SCH[schog2802]);

dc_overlapgroen(fc32, fc05, tog3205, fc11, tog3211, SCH[schog3205]);

dc_overlapgroen(fc34, fc08, tog3408, NG  , NG     , SCH[schog3408]);

dc_overlapgroen(fc36, fc05, tog3605, fc11, tog3611, SCH[schog3605]);

dc_overlapgroen(fc38, fc02, tog3802, NG  , NG     , SCH[schog3802]);

}

 

Functie:

 

/* ============================================================================================== */

/* Deelconflict overlapgroen. Het groen van een langzaamverkeerrichting in deelconflict wordt
 * vastgehouden in groen (RW[] BIT9) totdat de autorichting(en) een instelbare tijd groen is (zijn).
* Aanroepen in Synchronisaties_Add.
* Versie 20250401a
*
 * Voorbeeld: dc_overlapgroen(fc22, fc05, tog2205, fc11, tog2211, SCH[schog2205]);
*
 */

void dc_overlapgroen(count fclv,       /* deelconflict langzaam verkeer */
                         count fcauto1,    /* deelconflict auto 1           */
                         count toglvauto1, /* bij instelling T_max = 0 dan geen overlapgroen */
                         count fcauto2,    /* mag NG */
                         count toglvauto2), /* mag NG */
                         boolv  period)     /* extra voorwaarde                                   */
{

  RW[fclv] &= ~(BIT9);
 if (period)
{
  RT[toglvauto1] = (T_max[toglvauto1] > 0) && G[fclv] && RA[fcauto1];

  if (toglvauto2 > NG)
    { RT[toglvauto2] = (T_max[toglvauto2] > 0) && G[fclv] && RA[fcauto2]; }

  if (RT[toglvauto1] || T[toglvauto1] ||
      ((toglvauto2 > NG) && (RT[toglvauto2] || T[toglvauto2])))  RW[fclv] |= BIT9;
}
}                        

 

 

Ik heb BIT9 gebruikt, maar een andere BIT is misschien fraaier 😊

ccappendijk, 2026-01-28:
Overleg met Eindhoven eerst inplannen (Peter en Cyril)

mennowo, 2026-10-07:
@ccappendijk Wat is hiervan de status, nog steeds een wens? Zo ja, is er al code? Of kunnen we dit issue sluiten?

mennowo, 2026-11-06:
Dit verplaatsen we naar de backlog; het is nodig deze toevoeging uitgebreid te (duur)testen

closed 2026-11-06 by mennowo