mennowo/TLCGen #130: Extra elementen niet weg te halen ?

ccappendijk, 2025-05-20:
[image]

mennowo, 2025-05-20:
Gaat er iets mis met de update? Of is er iets dat na de update niet (meer) werkt (en dus met de update niets te maken heeft, maar veeleer met de nieuwe versie)? Om welke elementen gaat het? In exact welk werkblad van TLCGen? Is er een voorbeeld regeling? Exact welke knop doet het niet op exact welke plek?

ccappendijk, 2025-05-20:
[image]

[image]

[image]

mennowo, 2025-05-20:
Van willekeur is geen sprake. Het zal zo zijn dat het aan die knop verbonden commando soms wel en soms niet wordt geupdate, net afhankelijk van exact welke handelingen je uitvoert. Ik zal ernaar kijken.

closed 2025-05-22 by mennowo