mennowo/TLCGen #131: PTP Define niet beschikbaar in CCOL120?

ccappendijk, 2025-05-20:
[image]

LexTrafico, 2025-05-20:
CCOL ondersteunt max 4 PTP verbindingen. Is een oude afspraak met Astrin uit de tijd dat PTP koppelingen nog via RS232 liepen.
CCOL is daar nooit op aangepast, wellicht omdat de vraag ook nooit gesteld is. Vandaar de max 4 PTP_FOUT definities in cif.inc.

Ik moet ook zeggen dat ik er bij de PTP aanpassing ook 'blind' vanuit ben gegaan dat 4 het max zou zijn, en controleer daar in de code niet op.

Wel interessant dat Eindhoven meer dan 4 PTP verbindingne toepast. Ben benieuwd of dat in de praktijk ook goed uitpakt (aan de PB / container kant zullen ook buffers gereserveerd worden denk ik, 'k hoop dat dat goed gaat bij buiten-spec bedrijf).

Wellicht eens met Ton bespreken.
Lastige is dat het bij PTP over IP m.i. moet kunnen, maar bij PTP over RS232 zit je vast aan het max aantal beschikbare fysieke poorten en dat is (volgens de Astrin afspraak) 4. Vanuit CCOL is niet te controleren met welke PTP koppeling (IP of RS232) je van doen hebt, dus de programmeur moet zelf deze beperking in de smiezen houden.

ccappendijk, 2025-05-21:
Voorstel is om vanaf de 5de verbinding de compileerfout op 'te lossen' door een waarde in te vullen.

LexTrafico, 2025-05-21:
Zeker weten dat dat goed gaat? 

[image]

Overleg gehad met leveranciers, over PB en TLC besturing, ccol container, hun configuratie software, etc?

mennowo, 2025-05-22:
Mij lijkt dit ook een issue om eerst breder te bespreken. Het is hoe dan ook geen bug in TLCGen, maar eerder een functionele vraag gerelateerd aan CCOL en afspraken omtrent PTP koppelingen etc. Eigenlijk zou ik suggereren dat TLCGen het "goed" doet door "gewoon" voor de 5e verbinding de juiste code te genereren; vervolgens komt echter naar voren dat dit scenario niet wordt ondersteund door CCOL en aanverwant.

Ergo:

- Dit is volgens mij geen bug in TLCGen
- Desgewenst kunnen we het aantal PTP verbindingen binnen TLCGen maximeren op 4, zodat het aansluit op hetgeen functioneel wordt ondersteund door CCOL en aanverwant

Ik stel voor dit issue te sluiten.