mennowo/TLCGen #149: Wens + PR: geparametriseerde KAR richtingnummers instelbaar maken in GUI

LexTrafico, 2025-06-16:
Het is mogelijk om de richtingnummers waarmee KAR een aanvraag doet op een signaalgroep, via parameters te wijzigen (tabblad 'Algemeen' -> 'Prioriteit opties' het vinkje zetten bij 'KAR signaalgroep nummers in PRMs').
TLCGen maakt dan 'onder water', per KAR richting, parameters aan voor OV en voor hulpdiensten, bijvoorbeeld voor richting 05 zijn dat de PRMs karsg05=5 en karsghd05=5.

Op verzoek en met steun van BUKO heb ik een aanpassing gedaan in TLCGen zodat de waarde van deze PRM's ook in de GUI ingesteld kan worden. Dit is wenselijk voor het geval er bijvoorbeeld drie kruisingsvlakken geregeld worden met 05/105/205 of 105/205/305; in dat geval biedt het automatisch verlagen van richtingnummers boven de 200 geen soelaas.
Ook is het soms wenselijk om een tijdelijke regeling te maken met prio via KAR zonder dat de busmaatschappij voor de tijdelijke situatie de KAR database in de boardcomputers hoeft aan te passen (waarvoor de wegbeheerder ook de kruispuntconfiguratie in de KAR Geotool moet aanpassen). Door op voorhand de parameters in te stellen conform de 'oude' inrichting kan in sommige situaties de kruispuntconfiguratie tijdelijk wijzigen zonder dat aanpassingen in de boardcomputers nodig zijn.

Je zou deze aanpassingen natuurlijk in de tab.add kunnen doen, maar niet iedere VRA leverancier kan daar goed mee omgaan (bijvoorbeeld in FlowNodes en Eurocontrollers (Dynniq) is het problematisch wanneer het aantal elementen in de sys (h + add) afwijkt van het aantal elementen in de tab (c + add).
Vandaar de wens tot instelbaarheid in de GUI.

In de beoogde opzet komt een extra invulveld beschikbaar wanneer onder 'Algemeen' -> 'Prioriteit opties' het vinkje gezet wordt bij 'KAR signaalgroep nummers in PRMs'.
Voor OV is dat onderaan de ingreep configuratie onder 'Overig':

[image]

Voor HD is dat bovenaan de ingreep configuratie onder 'Opties - KAR': 

[image]

Standaard wordt het invulveld gevuld met de KAR richtingnummers die TLCGen zelf toepast; wanneer de gebruiker een andere waarde invult, wordt deze andere waarde toegepast voor de bijbehorende PRM karsg## of karsghd##.
De Spec Generator is hiervoor ook aangevuld.

Ik heb de aanpassing gedaan en zal die via een PR beschikbaar maken. Deze aanpassing is mede mogelijk gemaakt door BUKO.

mennowo, 2025-06-20:
Lijkt mij prima om over te nemen. Ik heb een paar opmerkingen geplaatst bij de PR. Met name waar het gaat om de logica gerelateerd aan de invul velden zou ik het graag nog bespreken. Kan graag ook even per telefoon als mijn review bij de PR vragen oproept!

LexTrafico, 2025-06-26:
Is met mijn aanpassingen / verklaring de behoefte aan telefonische bespreking nog steeds actueel? Zo ja, komt begin volgende week gelegen?

mennowo, 2025-06-28:
De geneste grid zou nog aangepast kunnen worden, is vooral een optisch ding.

Lastiger vind ik de logica in de setters voor de nummers; ik neig meer naar een opzet waarbij je gewoon invult wat je invult, en er bv een tekst bij staat wat daarmee gebeurt in de code gen. En je kunt je er ook een tekstveld bij maken met het 'resultaat' van de ingevulde waarde. Ik heb anders het idee dat dit wellicht vragen op kan roepen, wanneer men wat in wil vullen maar de logica verandert de ingevulde waarde.

mennowo, 2026-10-07:
@LexTrafico Is dit nog actueel? Zo ja, iets inplannen om gezamenlijk naar te kijken?