mennowo/TLCGen #157: Opmerkingen Marc Sherjon (nog opnemen hierin)

ccappendijk, 2025-07-05:


ccappendijk, 2025-07-06:
Als je toch bezig gaat:  Nog enkele zaken die mij zijn opgevallen:

 Na sommige wijzigingen blijft het ‘opslaan’ knopje ‘grijs’ ( greyed out )

Bij keuze Ccol 12.1 worden de simulatie ingangen van OV Prio NIET gegenereerd. ( HD wel )

Na aanpassing van het kruispuntplaatje, kwamen er 2 inputs op (bijna) ’t zelfde coördinaat.
Het 2e punt kan dan echter niet meer worden gedeselecteerd. Waardoor er in de dpl file extra ingangen worden aangemaakt.  ISMAX+1. Heb ik handmatig in de .tlc  file verwijderd.

Bij SIM van prio en HD ingangen wordt het FC nr ingevuld,  kan dat default op NG ? Hierdoor wordt alleen een ‘puls’ gegenereerd, i.p.v. ‘bezet’ houden tot richting groen is.

Zo ( eigenlijk ) ook voor drukknop ingangen

Sim file: =>  FC_nr[xx] = NG;

LexTrafico, 2025-07-07:
Het grijs blijven van de save button lijkt ook te gebeuren na het inklikken van (nieuwe) uitgangen in de bitmap en het wijzigen van fasen in de bitmap. Versie 12.4.0.16.

LexTrafico, 2025-07-08:
Ook bij het _verwijderen_ van een element via  de Gebruikersopties.
Bij toevoegen (via Gebruikersopties) wordt de knop 'opslaan' daarentegen wel actief.

ccappendijk, 2025-07-12:
Update aanvullingen door Cyril:

- Opslaan / knopjes grijs graag specificeren. Vraag is gesteld aan Menno. 
- CCOL12.1 simulatieingangen van OV Prio NIET gegenereerd. Is volgens mij al opgelost in 12.4.0.17. Ik heb het net gestest met een nieuwe programma en ik ziet de ingangen wel.
kruispuntplaatje is klein detail. Mag natuurlijk worden gezocht maar voor nu niet heel belangrijk.
- NG bij prioriteits ingangen + drukknoppen mag worden doorgevoerd. Geen prioriteit.
- .xml van Marc doet het niet meer.

closed 2025-07-12 by ccappendijk