mennowo/TLCGen #159: Prio: storing opvang lijkt niet te worden gegenereerd

LexTrafico, 2025-07-13:
Na het configureren van storings opvang van een prio (OV) in/uitmelding (in de boomstructuur als 'child' van een KAR melding) worden wel de elementen gegenereerd, maar niet de code voor in/uitmelden in <sys>prio.c. Versie 12.4.0.16.

mennowo, 2026-10-07:
@LexTrafico is dit nog actueel? Of is het verholpen?

mennowo, 2026-11-06:
De use case opvang storing in de vorm KAR > storing > opvangen met detectie wordt niet ondersteund. De code generator maakt enkel opvang storing code aan voor melding type "Detectoren"; de UI is hierop onlangs aangepast, zodat het vinkje "opvang storing" er enkel is bij type melding Detectoren, en voor de opvang is Detectoren de enige optie als type melding.

closed 2026-11-06 by mennowo