mennowo/TLCGen #160: pelotonkoppeling type 1 wordt niet / niet juist gegenereerd

LexTrafico, 2025-07-17:
Bij gebruik van pelotonkoppeling **type 1** met koppeling 'intern' aangevinkt wordt de code voor de betreffende koppeling: 

[image]

- niet gegenereerd wanneer er geen detector naam is opgegeven (maar die detectornaam zou niet nodig moeten zijn omdat een interne  koppeling is geselecteerd)
- foutief gegenereerd wanneer wel een detectornaam is opgenomen.

[image]

TLCGen versie 12.4.0.16.

mennowo, 2026-10-07:
@ccappendijk @LexTrafico Hebben we wellicht een voorbeeld van hoe het wél zou moeten zijn?

mennowo, 2026-11-06:
Dit is (volgens mij inclusief de opmerking van 2 augustus van @LexTrafico) verholpen met de merge van #235

closed 2026-11-06 by mennowo